nefarius/DsHidMini #259: Rumble (vibration) not working in DuckStation emu

Pyromanc3r, 2022-10-16:
Can anyone check if the vibration works in the DuckStation emulator with DS3 controller?
Previously I had ScpToolkit drivers and in the emulator settings the type of input Xinput was chosen and the vibration worked well.
I decided to try DsHidMini as a new modern solution for connecting DS3 running Win 10.
I deleted all the old drivers according to the instructions, cleaned the garbage with utilities "Legacinator" and "DriverStoreExplorer". In general, DsHidMini was installed normally, I have a genuine DS3 controller (CECHZC2R). The wired and wireless connection work perfectly. The only thing that vibration does not work in the DuckStation emulator. I even found here nearby "MightyHIDTest" utility and it gives out vibration with the SXS mode. I also tested the ePSXe 2.0.5 emulator in the game "Metal Gear Solid" with the "LilyPad plugin" and the vibration was present.
I don't know which side the DuckStation or DsHidMini problem is on which side? I tried to use different input modes in the emulator and in DSHMC.exe. Xinput, DirectinPut and etc. Different combinations, but the vibration in DuckStation has not earned (to clarify: in the control settings I use "Controller Type: Analog Controller (Dualshock)", Not a "Digital Controller"). I tried increase/decrease "Vibration Bias" but this did not give the effect.
I repeat that earlier vibration worked with SCP drivers (on the wire and through BT).
Can someone check DuckStation and DsHidMini together for vibration in games?

Windows 10 20H2, x64
BthPS3Setup_x64-v2.6.167 (setup-v2.6.174)
dshidmini_v2.2.282.0

Forusim, 2022-10-18:
For me rumble works for original DS3 (USB-attached) only with XInput.
Make sure to select XInput in DsHdiMini and Duckstation "Controller Backend" settings.

nefarius, 2022-10-19:
Works perfectly over here so IDK 😛

closed 2022-10-19 by nefarius