nefarius/DsHidMini #260: DSHMv3 does not allow setting LEDs states from config file

Kanuan, 2022-10-22:
# Context

In DS3's output report, there is:

- 1) 1 byte that sets LED states
- 2) 4 groups of 5 bytes, each one for each LED, that sets sets custom effects the related LED should execute
    - Effects are only executed if the related LED has been enabled via (1)

# Issue:

Currently, DSHMv3 allows settings (2), but not (1), so that means the LEDs are always off and no effects are shown

# Proposed solution:

1. Allow dshidmini to read from a property in the .json file called "LEDsState" that sets the corresponding LEDs states byte in the output report
2. Rename the "Enabled" properties in LEDsCustoms to "EffectsEnabled" to make things clearer.

[image]

nefarius, 2022-11-01:
@Kanuan FYI I renamed the suggested `LEDsEnabled` to `LEDFlags` and `Enabled` to `EnabledFlags`

Kanuan, 2022-11-07:
I might be doing something wrong, but the LED effects don't seem to be loading as expected.

1. I've connected the controller
2. Settings were already in custom LED settings, driver authority mode
3. Changed which LEDs were enabled to cause a hot-reload
4. This is what was sent to the controller:

[image]

In orange 1E means all LEDs are on (1e -> 11110), which is as expected.

The other 4 groups of 5 bytes are the effect related bytes:

- 1st byte -> Duration
- 2nd byte -> IntervalDuration
- 3rd byte -> EnableEffects (0x10 -> ON // 0x00 -> off)
- 4th byte -> Interval... ON?
- 5th byte -> Interval... OFF?

Below are my [DsHidMini.json](https://github.com/ViGEm/DsHidMini/files/9947826/DsHidMini.txt) LED settings:

[image]

-  By my settings, the 2nd byte should've been 0x0F, and the 4th and 5th bytes that are 0xFF in the output report should've been 0x7F
- I know the values I've used should be working because that's what is being used in DS4 lightbar translation to have a blinking LED, as shown in the image below:

[image]

Kanuan, 2022-11-07:
Oh, and besides that, when "cold-loading" the settings in custom LED mode all LEDs remain off until a hot-reload occurs

nefarius, 2022-11-07:
> The other 4 groups of 5 bytes are the effect related bytes:
> 
> * 1st byte -> Duration
> * 2nd byte -> IntervalDuration
> * 3rd byte -> EnableEffects (0x10 -> ON // 0x00 -> off)
> * 4th byte -> Interval... ON?
> * 5th byte -> Interval... OFF?

You got byte 4 and 5 mixed up btw.:

```c
typedef struct _DS_LED
{
	UCHAR Duration;

	UCHAR IntervalDuration;

	UCHAR EnabledFlags;

	UCHAR IntervalPortionOff;

	UCHAR IntervalPortionOn;

} DS_LED, * PDS_LED;
```

Kanuan, 2022-11-07:
On my .json settings the order is correct, but I'm setting the same value on both of them though, so it should work.

Kanuan, 2022-11-07:
Sorry. I took the screenshot and then noticed the authority parameter was missing. I re-did the tests with the parameter then and got the same results but forgot to update the screenshot

nefarius, 2022-11-07:
Understood, will investigate on my end.

Kanuan, 2024-04-24:
Fixed in #348

commit e49bbac0 (2022-11-01): Resolves #260
Implemented reading LEDFlags from config
Renamed Enabled to EnabledFlags
closed 2024-04-24 by Kanuan