nefarius/DsHidMini #263: High CPU of Windows Driver Foundation after a while in SXS mode

FuJa0815, 2022-11-13:
- [x] I have searched open and closed issues for duplicates
- [x] I am submitting a bug report for existing functionality that does not work as intended
- [x] I have read https://vigem.org/Community-Support/
- [x] This isn't a feature request or a discussion topic

----------------------------------------

### Bug description
I am using DsHidMini for a Dualshock 3 controller. After a while the CPU usage of `Windows Driver Foundation - User-mode Driver Framework Host Process` starts rising and rising. This leads to PCSX2 getting slower and slower. After reconnecting the controller the CPU usage is down again and everything runs smoothly. I have already reinstalled the drivers, did not work.

### Screenshots

[image]
[image]

### Machine info

**CPU-Architecture:** x64
**Windows version:** Windows 10.0.19043 Build 19043
**Software/driver version(s):** 2.2.282.0

### Any other helpful information
[DSHMC20221113.txt](https://github.com/ViGEm/DsHidMini/files/9997467/DSHMC20221113.txt)

I am aware that reproducing this error might be difficult. I also don't have any "proof" that DsHidMini is 100% at fault here, but reconnecting the controller fixes this issue again for a while, therefore it seems obvious to me.

nefarius, 2022-11-13:
Are you using the XI proxy DLL?

mrpenguinb, 2022-11-22:
I have the same issue, after a while the WDF process got to 10 or more percent CPU usage. I'm using the x64 XInput .dll for PCSX2.

englebert3rd, 2022-12-27:
I'm getting the same issue using the proxy DLL.

nefarius, 2024-05-24:
To those affected feel free to [test it with the complete overhaul of the DLL](https://docs.nefarius.at/projects/DsHidMini/SCP-XInput-Bridge/#latest-downloads), it works with the old driver just fine, no need for any other upgrades.

MoaMission, 2024-05-24:
I can confirm that the issue seems to be resolved. I haven't noticed a VRAM or CPU ramp-up after playing for ten minutes like before.

Thanks so much for addressing this!

nefarius, 2024-05-28:
Pretty much a complete rewrite, yeah. More sophisticated and thought through. You can see all the work in the finished pull requests and the commit history.

closed 2024-07-03 by nefarius

dotur911, 2026-08-25:
I can can confirm ,i have that problem too.

MoaMission, 2026-08-25:
I remember this post from May. I have a suspicion it's because that very particular project uses an outdated emulator / outdated methods as a baseline for playing Twilight Princess. You should really take it up with the devs of that 'port' program...not here.

...but that's just my take. I'll let nefarius respond and look into it.

nefarius, 2026-08-25:
OK so since it is apparently too hard to grasp that this topic is done and had nothing to do with this project I'll lock this issue.