nefarius/DsHidMini #276: XInput-compatible devices does not work properly with some applications

gandalfthewhite19890404, 2023-01-15:
### Bug description
Native XInput-compatible devices does not work properly with some applications after DsHidMini installed. For example Xenia, it is a x360 console emulator. 
- When Xnput mode in DsHidMini enabled - only DS3 pad is working, it on Xinput port 0? Can we change ports for newly added DS3 pads? But something tells me if we are set it to 1,2 or 3 we will get something like below.
- When other modes are enabled - native XInput pad does not work at all with Xenia.

So I had to completely remove DsHidMini and installed ScpToolkit - it is works good.

### Machine info

**CPU-Architecture:**  x64
**Windows version:** Windows 10 22H2
**Software/driver version(s):** 2.2.282.0

nefarius, 2023-01-15:
Not a bug, read the documentation.

closed 2023-01-15 by nefarius

nefarius, 2023-01-15:
Also use the search next time, see #40