nefarius/DsHidMini #277: Bluetooth "Driver Entry Point Not Found"

Pilmyi, 2023-01-16:
- [x] I have searched open and closed issues for duplicates
- [x] I am submitting a bug report for existing functionality that does not work as intended
- [x] I have read https://vigem.org/Community-Support/
- [x] This isn't a feature request or a discussion topic

----------------------------------------

### Bug description
Describe here the issue that you are experiencing.

Uninstalled this program yet my Bluetooth seems to be not working anymore. Searched and followed almost a dozen other solutions for similar issues (yet can't quite find the same exact one). I get the Bluetooth yellow triangle error symbol in Device Manager with Device Status as:

"Windows cannot load the device driver for this hardware. The driver may be corrupted or missing. (Code 39)

{Driver Entry Point Not Found}
The %hs device driver could not locate the entry point %hs in driver %hs."

I've attempted uninstalling drivers, rebooting, manually installing drivers,etc. nothing works. This is a Dell Latitude notebook running Windows 10, using build in Bluetooth. Really can't afford a factory reset right now all over this one program. Please help, thank you.

### Steps to reproduce
- using hyphens as bullet points
- list the steps
- that reproduce the bug

-Uninstall
-Attempt to use Bluetooth?

**Actual result:** Describe here what happens after you run the steps above (i.e. the buggy behaviour)

Bluetooth is gone, drivers can't be reinstalled correctly

**Expected result:** Describe here what should happen after you run the steps above (i.e. what would be the correct behaviour)

Bluetooth should work as normal

### Screenshots

### Machine info

**CPU-Architecture:** x86, x64, ARM, ARM64, ...
**Windows version:** Windows 10 20H2
**Software/driver version(s):** 1.4.14

x64 Win 10

### Any other helpful information

nefarius, 2023-01-16:
> Searched and followed almost a dozen other solutions for similar issues (yet can't quite find the same exact one)

Except the [official docs](https://vigem.org/projects/BthPS3/Frequently-Asked-Questions/) it seams...

Also [run this](https://github.com/nefarius/Legacinator).

Cheers

closed 2023-01-16 by nefarius

Pilmyi, 2023-01-20:
Thanks, but when I try to run the Legacinator I get a "Legacinator has stopped working" and it crashes before anything can happen