nefarius/DsHidMini #294: Bug when pressing 2 Buttons (Circle + R2)

StrawFox64, 2023-05-28:
- [X] I have searched open and closed issues for duplicates
- [X] I am submitting a bug report for existing functionality that does not work as intended
- [X] I have read https://vigem.org/Community-Support/
- [X] This isn't a feature request or a discussion topic

----------------------------------------

### Bug description
I was playing rocket league on the ds3 with the newly installed driver. but I could see that in some moments that I used the accelerator (R2) and then the boost (Circle) the game camera was buggy and kept pulling to the side, it was also deactivating and activating the circle. The bug only stopped if I released all the buttons, but then it came back. To make sure it wasn't the control, I tested it on my ps3 in the old ds3 drivers the scptoolkit and it worked perfectly

### Steps to reproduce
Described in the tab above

**Actual result:** Described in the tab above (Bug Description)
**Expected result:** Described in the tab above (Bug Description)

### Screenshots

### Machine info

**CPU-Architecture:** x86, x64
**Windows version:** Windows 10 22H2
**Software/driver version(s):** v2.2.282 

### Any other helpful information

nefarius, 2023-05-28:
What mode are you using? The driver simply translates what it receives from the hardware, this sounds like an issue with whatever button layout you are using. [See documentation](https://vigem.org/projects/DsHidMini/HID-Device-Modes-Explained/).

StrawFox64, 2023-05-28:
I was using the drive in xinput mode

nefarius, 2023-05-29:
Well, try DS4Windows mode, I bet everything works there. As explained in the docs, there is no guarantee XI mode will be picked up correctly in modern games.

closed 2023-05-29 by nefarius