nefarius/DsHidMini #310: DsHidMini v3 not loading device specific settings when controller is connected via BT

Kanuan, 2023-08-13:
This is the configuration file I'm using: [DsHidMini.txt](https://github.com/nefarius/DsHidMini/files/12328893/DsHidMini.txt) (changed from .json to .txt to upload the file here on github)

This is the tracing results when connecting the controller via USB:

[image]

This is the result when connecting via BT:

[image]

----------- 

I'm pretty sure I wasn't running on this problem before updating the BthPS3 version (currently I'm using version 2.8.320.0). Will test if it's actually a BthPS3 issue somehow

Kanuan, 2023-08-13:
> I'm pretty sure I wasn't running on this problem before updating the BthPS3 version (currently I'm using version 2.8.320.0). Will test if it's actually a BthPS3 issue somehow

Yeah, no,  the same issue happens under BthPS3 v2.6.127.0. I think I just didn't notice this issue before because I was using global settings the majority of time

nefarius, 2023-08-13:
Not sure I ever tested it wireless since it was introduced, so can totally be the case it's overlooked, will check it out.

Kanuan, 2024-04-16:
In `configuration.c`, the `ConfigLoadForDevice` method gets the string that represents the device MAC address from the `Context->DeviceAddressString` field. The issue is that this field is not set when the device is connected via BT.

I've added a tracing message for it to show this field value if it fails to load the settings and it returns blank when via BT:
[image]
[image]

From what I've looked around, The device's `DeviceAddressString` field is only set in the `DsUdb_PrepareHardware` method of the `DsUsb.c` file, but this method is only called if the device is connected via USB, therefore the `DeviceAddressString` field is never set via BT:
[image]
[image]

I tried to set `DeviceAddressString` in the `DsDevice_ReadProperties` method of the `device.c` file immediately after it sets the `DeviceAddress` field. This works in the sense that the field is correctly set, but for some reason putting this piece of code here causes the driver to get stuck immediately after it:
[image]
[image]

Kanuan, 2024-04-16:
From the traceview log, the driver hangs if  `DeviceAddressString` is set before invoking the `DsDevice_RegisterBthDisconnectListener` method. Setting the `DeviceAddressString` field AFTER both listerner-related methods are invoked fix the hanging and the code works as expected, allowing the driver to properly load device-specific configurations.

I'm really curious to why the driver hangs if the `DeviceAddressString` field is set before the listener-related methods are invoked, nothing in them seems to make use of it, but regardless, this got the "works on my machine" seal of approval. Will prepare a PR for the fix.

Kanuan, 2024-04-16:
From the traceview log, the driver hangs if  `DeviceAddressString` is set before invoking the `DsDevice_RegisterBthDisconnectListener` method. Setting the `DeviceAddressString` field AFTER both listerner-related methods are invoked fix the hanging and the code works as expected, allowing the driver to properly load device-specific configurations.

[image]

I'm really curious to why the driver hangs if the `DeviceAddressString` field is set before the listener-related methods are invoked, nothing in them seems to make use of it, but regardless, this got the "works on my machine" seal of approval. Will prepare a PR for the fix.

nefarius, 2024-04-16:
If the tracing just stops and the driver responding, it means it crashed. Look in the event viewer for Critical or Error events from the user-mode driver reflector service, I bet you find one or more crash logs.

nefarius, 2024-04-16:
Should look similar to #333

Kanuan, 2024-04-16:
[image]

nefarius, 2024-04-16:
Yeah that's why I thought, we're probably reading or writing invalid/uninitialized memory somewhere during this exact sequence or in parallel and it goes poof. WinDbg session or code review could uncover why exactly.

nefarius, 2024-04-16:
Once you make a PR and I see the diff, maybe something obvious jumps out.

PR #341 (2024-04-16): Fix loading device-specific configuration when it's connected via bluetooth
Fixes #310
closed 2024-04-16 by nefarius