nefarius/DsHidMini #311: Controller error if via cable after waking system from sleep

Kanuan, 2023-08-21:
If controller is connected via cable while waking system from sleep, then the device will be stuck in a error state until the controller is reconnected.

[image]

nefarius, 2024-05-26:
@Kanuan could you possibly try to replicate that while also having TraceView running before, during and after resume and share if you get anything captured?

nefarius, 2024-05-26:
Got one:

```text
1: kd> !analyze -v
*******************************************************************************
*                                                                             *
*                        Bugcheck Analysis                                    *
*                                                                             *
*******************************************************************************

Unknown bugcheck code (0)
Unknown bugcheck description
Arguments:
Arg1: 0000000000000000
Arg2: 0000000000000000
Arg3: 0000000000000000
Arg4: 0000000000000000

Debugging Details:
------------------

KEY_VALUES_STRING: 1

    Key  : Analysis.CPU.mSec
    Value: 3124

    Key  : Analysis.Elapsed.mSec
    Value: 3410

    Key  : Analysis.IO.Other.Mb
    Value: 3

    Key  : Analysis.IO.Read.Mb
    Value: 5

    Key  : Analysis.IO.Write.Mb
    Value: 4

    Key  : Analysis.Init.CPU.mSec
    Value: 17109

    Key  : Analysis.Init.Elapsed.mSec
    Value: 729965

    Key  : Analysis.Memory.CommitPeak.Mb
    Value: 221

    Key  : Failure.Bucket
    Value: 0x0_KERNELBASE!wil::details::DebugBreak

    Key  : Failure.Hash
    Value: {bf435393-ab1c-2b67-8d0c-a2c789a315b3}

    Key  : Hypervisor.Enlightenments.Value
    Value: 0

    Key  : Hypervisor.Enlightenments.ValueHex
    Value: 0

    Key  : Hypervisor.Flags.AnyHypervisorPresent
    Value: 0

    Key  : Hypervisor.Flags.ApicEnlightened
    Value: 0

    Key  : Hypervisor.Flags.ApicVirtualizationAvailable
    Value: 0

    Key  : Hypervisor.Flags.AsyncMemoryHint
    Value: 0
[... 335 lines cut ...]
DRV - 234281: DMF_Generic_ModuleSelfManagedIoFlush -  [DMF_Generic_ModuleSelfManagedIoFlush] <-- Exit <dmfObject=0x0000029024EC51A0 [DsHidMini] returnValue=0>
DRV - 234282: DMF_Generic_ModuleSelfManagedIoFlush -  [DMF_Generic_ModuleSelfManagedIoFlush] --> Entry <DmfModule=0xFFFFFD6FDB132F18 dmfObject=0x0000029024ECD190 [HashTable]
DRV - 234283: DMF_Generic_ModuleSelfManagedIoFlush -  [DMF_Generic_ModuleSelfManagedIoFlush] <-- Exit <dmfObject=0x0000029024ECD190 [HashTable] returnValue=0>
DRV - 234284: DMF_Generic_ModuleSelfManagedIoFlush -  [DMF_Generic_ModuleSelfManagedIoFlush] --> Entry <DmfModule=0xFFFFFD6FDB1338C8 dmfObject=0x0000029024ECC7E0 [VirtualHidMini]
DRV - 234285: DMF_Generic_ModuleSelfManagedIoFlush -  [DMF_Generic_ModuleSelfManagedIoFlush] <-- Exit <dmfObject=0x0000029024ECC7E0 [VirtualHidMini] returnValue=0>
DRV - 234286: DMF_Bridge_ModuleSelfManagedIoFlush -  [DMF_Bridge_ModuleSelfManagedIoFlush] --> Entry
DRV - 234287: Bridge_DefaultEvtDeviceSelfManagedIoFlush -  [Bridge_DefaultEvtDeviceSelfManagedIoFlush] --> Entry
DRV - 234288: Bridge_DefaultEvtDeviceSelfManagedIoFlush -  [Bridge_DefaultEvtDeviceSelfManagedIoFlush] <-- Exit
DRV - 234289: DMF_Bridge_ModuleSelfManagedIoFlush -  [DMF_Bridge_ModuleSelfManagedIoFlush] <-- Exit
DRV - 234290: DMF_ModuleCollectionDispatchVoid -  [DMF_ModuleCollectionDispatchVoid] <-- Exit <ModuleCollectionHandle=0x0000029024EC4850>
DRV - 234291: DMF_ModuleCollectionSelfManagedIoFlush -  [DMF_ModuleCollectionSelfManagedIoFlush] <-- Exit <DmfCollection=0xFFFFFD6FDB13B858>
DRV - 234292: DmfContainerEvtDeviceSelfManagedIoFlush -  [DmfContainerEvtDeviceSelfManagedIoFlush] <-- Exit
FX  - 234293: FxSelfManagedIoMachine::ProcessEvent - WDFDEVICE 0xFFFFFD6FDB14B698 !devobj 0x0000029024EBB2E0 entering self managed io state FxSelfManagedIoFlushed from FxSelfManagedIoFlushing
FX  - 234294: FxPkgPnp::NotPowerPolicyOwnerEnterNewState - WDFDEVICE 0xFFFFFD6FDB14B698 !devobj 0x0000029024EBB2E0 entering not power policy owner state WdfDevStatePwrPolStopped from WdfDevStatePwrPolStoppingSendStatus
FX  - 234295: FxPkgPnp::NotPowerPolicyOwnerEnterNewState - WDFDEVICE 0xFFFFFD6FDB14B698 !devobj 0x0000029024EBB2E0 entering not power policy owner state WdfDevStatePwrPolRemoved from WdfDevStatePwrPolStopped
FX  - 234296: FxPkgPnp::PnpEnterNewState - WDFDEVICE 0xFFFFFD6FDB14B698 !devobj 0x0000029024EBB2E0 entering PnP State WdfDevStatePnpFailedPowerPolicyRemoved from WdfDevStatePnpFailed
FX  - 234297: FxPkgPnp::PnpEnterNewState - WDFDEVICE 0xFFFFFD6FDB14B698 !devobj 0x0000029024EBB2E0 entering PnP State WdfDevStatePnpFailedWaitForRemove from WdfDevStatePnpFailedPowerPolicyRemoved
FX  - 234298: FxPkgPnp::PowerEnterNewState - WDFDEVICE 0xFFFFFD6FDB14B698 !devobj 0x0000029024EBB2E0 entering Power State WdfDevStatePowerStopped from WdfDevStatePowerPowerFailedPowerDown
DRV - 234299: DsUsb_EvtUsbInterruptPipeReadComplete -  [DsUsb_EvtUsbInterruptPipeReadComplete] --> Entry
DRV - 234300: DumpAsHex -  [DumpAsHex]>> USB - Buffer length: 0049, buffer content: 0100000000007E807E7E00000000000000000000000000000000000000030516FFB30007330077004001EF01E701930005
DRV - 234301: Ds_ProcessHidInputReport -  [Ds_ProcessHidInputReport] --> Entry
DRV - 234302: DMF_VirtualHidMini_InputReportGenerate -  [DMF_VirtualHidMini_InputReportGenerate] --> Entry
---- end of merged log ----
32 error events encountered in logs
```

nefarius, 2024-05-26:
So looks like that after power-up we successfully start to receive and process HID input reports again, however we hit an invalid state of the VirtualHid DMF module:

```text
1: kd> kc
 # Call Site
00 KERNELBASE!wil::details::DebugBreak
01 dshidmini!DMF_HandleValidate_IsOpened
02 dshidmini!DMF_HandleValidate_ModuleMethod
03 dshidmini!DMF_VirtualHidMini_InputReportGenerate
04 dshidmini!Ds_ProcessHidInputReport
05 dshidmini!DsUsb_EvtUsbInterruptPipeReadComplete
06 WUDFx02000!FxUsbPipeContinuousReader::_FxUsbPipeRequestComplete
07 WUDFx02000!FxRequestBase::CompleteSubmitted
08 WUDFx02000!FxIoTarget::CompleteRequest
09 WUDFx02000!FxIoTarget::RequestCompletionRoutine
0a WUDFx02000!FxIoTarget::_RequestCompletionRoutine
0b WUDFHost!CWudfIoStack::OnCompletion
0c WUDFHost!CWudfIrp::CompleteRequest
0d WUDFHost!CWudfIrpT<CWudfIoIrp,IWudfIoIrp2,_WUDFMESSAGE_IO_HEADER *,_WUDFMESSAGE_REPLY_HEADER *>::CompleteRequest
0e WUDFHost!CWudfIoDispatcher::_CompleteIrp
0f WUDFHost!CWudfIoDispatcher::_TryCompleteIrp
10 WUDFHost!CWudfIoDispatcher::_IoCompletedWorker
11 WUDFHost!CWudfIoDispatcher::IoCompletionRoutine
12 ntdll!RtlpTpIoCallback
13 ntdll!TppWorkerThread
14 KERNEL32!BaseThreadInitThunk
15 ntdll!RtlUserThreadStart
```

```c

VOID
DMF_HandleValidate_IsOpened(
    _In_ DMF_OBJECT* DmfObject
    )
/*++

Routine Description:

    Verifies that a DMF_OBJECT Module State is Opened.

    NOTE: This call is only meant for debug purposes.

Arguments:

    DmfObject - The DMF Object to validate.

Return Value:

    None. Failure causes an assert to trigger.

--*/
{
    UNREFERENCED_PARAMETER(DmfObject);

    DmfAssert(DmfObject != NULL);
    DmfAssert(DMF_OBJECT_SIGNATURE == DmfObject->Signature);
    DmfAssert(ModuleState_Opened == DmfObject->ModuleState);
}
```

closed 2026-09-05 by nefarius