nefarius/DsHidMini #314: Fix name scheme for LED effects customs in the driver

Kanuan, 2023-08-25:
Apparently this is incorrect for how LED effects work:
```
1st byte -> Duration
2nd byte -> IntervalDuration
3rd byte -> EnableEffects (0x10 -> ON // 0x00 -> off)
4th byte -> Interval OFF
5th byte -> Interval ON
```

The correct seems to be:

```
- 1st byte: Duration
    - 0 = effect is off
    - 0xFE = effect runs for ~5s then LED is turned off
    - 0xFF = effect repeats forever
- 2nd AND 3rd byte: Cycle duration
    - Cycle duration is probably (ClockTicks * tick duration) 
- 4th byte: Off period cycles
- 5th byte: On period cycles
```

The effect the goes like this:
```
OffPeriodCycles * CycleDuration + OffPeriodCycles * CycleDuration
```

Example: by setting everything to the max (0xFF) we have:

1. LED starts off and stays off for ~16s
    - (OffPeriodCycles * CycleDuration) = (0xFF * 0xFFFF) 
2. LED turns ON for ~16s
    - (OffPeriodCycles * CycleDuration) = (0xFF * 0xFFFF)
3. Repeat forever

Based on this, at max cycle duration (0xFFFF), the On/Off period durations will be ~1s for 0x10,  ~2s to 0x20 and so forth.

Based on this, I recommend renaming dshidmini driver's fields related to led customs to:

```
1st byte -> Duration
2nd byte -> TicksPerCycle1
3rd byte -> TickesPerCycle0
4th byte -> Off_IntervalCycles
5th byte -> On_IntervalCycles
```

Kanuan, 2023-08-25:
I'm assuming it's counting clock ticks. Since I'm not sure maybe "ticks" should be replaces by something else

Kanuan, 2024-04-20:
I think this would be more accurate:

```
1st byte -> Duration
2nd byte -> BasePeriod1
3rd byte -> BasePeriod0
4th byte -> OffPeriods
5th byte -> OnPeriods
```

nefarius, 2024-04-24:
This can be considered done, right?

Kanuan, 2024-04-24:
Yes!

closed 2024-04-24 by Kanuan