nefarius/DsHidMini #320: Windows remembers HID type after changing between modes

Kanuan, 2023-09-07:
# The issue

DsHidMini has different HID modes, and each HID mode can have its own Hid type. Currently, the 2 types used are Vendor-Defined for DS4W mode and Game-compliant for the others. The issue happens when Windows "remembers" the HID type of a previous HID mode and uses it for other modes

# Reproducing the issue

The issue can be consistently reproduced with DsHidMini v3 via BT (USB not tested):

1. Connect Controller via BT
2. Set controller to Global mode
    - Profile/Custom mode currently do not work via BT in DsHidMini v3, hence global mode is necessary
3. Uninstall the "PS3 gamepad" via Device Manager
    - This is so Windows "forgets" all saved information about the controller
4. Configure a Profile to use DS4W mode, then set this profile as Global
    - This is necessary so the DS4W mode is the first mode the controller is set on 1st connection
5. Connect the controller. It should load the global profile in DS4W mode
6. Confirm the controller is "Vendor-defined" on device manager
7. Change the global profile to SXS or SDF mode
8. Reconnect the controller so it changes mode
    - May require 2 reconnections
9. Confirm the controller is in SXS/SDF mode
10. Confirm on device manager the type of controller
    - Expected: HID compliant game controller
    - Result: Controller still as HID Vendor-Defined
    
[image]

# Workaround for the issue

Uninstalling the "PS3 gamepad" device via Device Manager makes Windows forget the last used HID type, then it will get stuck again on the first mode on the next "1st" connection

nefarius, 2023-09-07:
Maybe we need some registry purging on mode change that the control app can do transparently. I'll see what we can do. Maybe calling the device uninstall instead of port cycle is sufficient.

Kanuan, 2023-09-08:
Apparently, this is just a descriptive name. We are still not sure if there is a need to re-setup the device on HID mode change, but in practical terms all HID modes work as expected.

The only known issue we are currently aware of that is derived from this behavior is DS4Windows ignoring the DS3 in SXS mode if its HID description is "vendor-defined". But in this specific case it's actually an issue on DS4Windows side.

DS4Windows (as of v3.2.14) literally compares the device HID description with ~"HID vendor defined" and ignores it if it matches, but this is not the proper way of checking for a vendor-defined device since the description itself is translated if the PC is in another language and will not match what DS4Windows is checking for.

nefarius, 2024-05-18:
Not really resolved but not a big deal and doesn't warrant the effort to address.

closed 2024-05-18 by nefarius