nefarius/DsHidMini #321: Rework device and host MAC address discovery and pairing

nefarius, 2023-09-25:
Currently the host (if any) and device MAC address are requested on device boot and fails it entirely, if they do not succeed.

This should be converted from a hard error to a "warning" with fallback actions. This would improve the following situations:

- Fake controllers that are timing-sensitive might start working if we do not or simply send these requests later during runtime
- We can support aftermarket controllers without MAC address reporting like the Retro Fighters Defender
- We can add a mechanism to read or re-pair Bluetooth information during runtime instead of once on boot

nefarius, 2024-05-21:
One more discovery thanks to ps3 packet captures.

It looks like the console sends the initial output report with the led state over the control endpoint. Subsequent output reports, like rumble, are sent via interrupt out. We should make the driver do exactly that as well, plus cross check with the captures, how much time passes between each request between ps3 and DS3.

nefarius, 2024-05-21:
The Ep0 report is also slightly shorter and doesn't use a report id.

3asba-3asba, 2024-07-01:
it still dont work im using a ps3 controller original and the pairing comes out as failed

3asba-3asba, 2024-07-01:
its weirder even 
[image]
it doesnt even give answer on both sections

[image]
is this suppose to b like this ? 

and the controller runs fine on testing 
[image]

closed 2026-09-06 by nefarius