nefarius/DsHidMini #326: Smallest issue ever, ONE button giving me grief with PCSX2

supersaiyz, 2023-11-01:
- [x ] I have searched open and closed issues for duplicates
- [ x] I am submitting a bug report for existing functionality that does not work as intended
- [x ] I have read [the project documentation](https://docs.nefarius.at/projects/) and not found a solution to my problem there
- [x ] This isn't a feature request or a discussion topic

----------------------------------------

### Bug description

Up button on the directional pad is not functioning properly when used for menu selection in the game Bujingai - The Forsaken City emulated on PCSX2

### Steps to reproduce

Pressing up on the directional pad at a menu screen (the title screen, option screen, tutorial screen, etc)

**Actual result:**  Instead of moving up one menu item selection, menu selection will scroll DOWN as if the down button on the directional pad is being held.  Pressing the up button again does nothing, and while no other button press has any effect, actually pressing the down button one time stops the repeated scrolling.
**Expected result:** The correct behavior would be the moving of the menu cursor or selected menu item moving up one item.

### Screenshots

### Machine info

I don't have my exact specs atm 
**CPU-Architecture:** x86, x64, ARM, ARM64, ...
**Latest Windows
**Sony Vaio 2023 intel core i5

### Any other helpful information
Not a game breaker, and the menus in this game in particular go back to the top selection after passing the bottom most selection.  I haven't tested it on any other games with PSCX2, I've just started using it.  Could it be an anti-piracy measure from the game software?

nefarius, 2023-11-01:
Not a DsHidMini topic.

closed 2023-11-01 by nefarius

supersaiyz, 2023-11-01:
erp.  I thought it was since I'm using dshidmini to use my DS3 with PCXS2.  Anyway to confirm, from the PCXS2 dashboard the d-pad works correctly, but it does not work correctly in-game with Bujingai.  Also, change your avatar you clown-ass dickhole

nefarius, 2023-11-01:
Change your attitude, child 🙂