nefarius/DsHidMini #330: Plugging controller in results in showing as wireless and wired device simultaneously

vertigo220, 2023-11-19:
- [x ] I have searched open and closed issues for duplicates
- [ x] I am submitting a bug report for existing functionality that does not work as intended
- [x ] I have read [the project documentation](https://docs.nefarius.at/projects/) and not found a solution to my problem there
- [ x] This isn't a feature request or a discussion topic

----------------------------------------

### Bug description

I was connected via Bluetooth but the battery was getting low, so I plugged it in. After doing so, it was not seen/recognized by Epic Games/Fortnite, but Steam still saw it, so I knew it was connected and "working." After some troubleshooting, I realized it was showing twice, once under wireless/BT and once wired, both with the same ID. I'm guessing Steam is somehow able to deal with the fact it was basically two devices simultaneously while Epic wasn't. DsHidMini (or maybe it's an issue with BthPS3) should remove it as a wired/wireless device when it connects as the other.

### Steps to reproduce

- Connect a DS3 controller wirelessly
- Plug it in to connect it wired
- See that it's showing twice, one wireless and one wired

**Actual result:** See above
**Expected result:** See above

### Screenshots

### Machine info

**CPU-Architecture:** x64
**Windows version:** Windows 10 22H2
**Software/driver version(s):** 2.2.282

### Any other helpful information

nefarius, 2023-11-20:
Kinda know bug, not sure it can ever be fixed properly the way Windows works.

nefarius, 2023-11-20:
I thought I had a working mechanism but oh well.

closed 2026-09-05 by nefarius