nefarius/DsHidMini #332: [Windows 10] Controller disappears from Device Manager and DSHMC

AFriendlyIrin, 2023-11-28:
### Bug description

Everything was working perfectly, but all of a sudden my controller can no longer be detected by Device Manager or DSHMC, whether by USB or wireless. The Nefarius HID Devices category in Device Manager has completely disappeared. However, I can still find the drivers with Driver Store Explorer, and attempting to remove dshidmini.inf produces an error stating that it is still in use. I attempted to delete and reinstall the drivers as well as restart my computer several times, but the problem persists.

The Bluetooth drivers are also installed.

The controller LED is a solid light at position 4. Plugging/unplugging through USB doesn't reset this.

### Steps to reproduce

I am not certain how to reproduce the issue, as I can't get DsHidMini working again to test it.

### Screenshots

[image]
[image]
[image]

### Machine info

**CPU-Architecture:** x64
**Windows version:** Windows 10.0.19045
**Software/driver version(s):** 2.2.282

nefarius, 2023-11-28:
What is the error code and message of the device with the exclamation mark?

nefarius, 2023-11-28:
This error [is covered in the FAQ](https://docs.nefarius.at/projects/DsHidMini/FAQ/#how-to-fix-device-timeout).

AFriendlyIrin, 2023-11-28:
It was working just a few minutes earlier, though, and I've been using this controller for years without issue. How did the hardware suddenly break?

nefarius, 2023-11-28:
> It was working just a few minutes earlier, though, and I've been using this controller for years without issue. How did the hardware suddenly break?

IDK, I never could reproduce this myself, I have no solution for you.

AFriendlyIrin, 2023-11-28:
I switched back to ScpToolkit and it works fine again.

You might want to revise that bit in your FAQ about how DsHidMini is so much better than your old decrepit version.

nefarius, 2023-11-28:
> I switched back to ScpToolkit and it works fine again, so it's definitely a software problem.
> 
> You might want to revise that bit in your FAQ about how DsHidMini is so much better than your old decrepit version.

Or, how about, people start contributing fixes and solutions instead of waiting for me to do all the work? 😉

nefarius, 2023-11-28:
> You might want to revise that bit in your FAQ about how DsHidMini is so much better than your old decrepit version.

As I said; I do not have any issues. Come up with the solution, the code is open, expecting a fix next week! It's easy, it's just software, after all!

nefarius, 2023-11-28:
> I switched back to ScpToolkit and it works fine again

NVM then, glad to see it resolved! ♥

closed 2023-11-28 by nefarius