nefarius/DsHidMini #357: Only update device context's host address by using its respective GetFeature request

Kanuan, 2024-04-28:
When on USB, there are 2 situations in which DsHidMini updates the device's host address field that is stored in the Device Context:

1. It sends a GetFeature request and saves the host address informed by the controller when it responds
    - Occurs in the `DsUsb_PrepareHardware` method
2. When a pairing request is sent to the controller and is considered successful, then it automatically updates the device's context host address field to the same one sent in the request
    - Occurs in the method currently known as `DsUsb_Ds3PairToFirstRadio`
    
I'd like to know your thoughts on removing the automatic updating done in (2), and replacing it with what happens in (1).

My line of thinking is that it would be better to always confirm what host address the controller is informing instead of assuming it was updated by the pairing request.

This would give us more info in situations in which the driver considers the pairing request succeeded but the controller continues informing it is paired to another MAC address

nefarius, 2024-04-29:
Agreed, the SET request being successful *may* not infer that the address was written properly, so better to do:
- Set new address to device
- Query the address from device
- If it matches with what we sent, update it in the context

Also another remark; IIRC there currently is two places for either device or host address where it is stored as a byte array/struct and a string, this makes it susceptible to bugs where only one is updated, we should look into streamlining that as well.

Kanuan, 2024-04-30:
If the request to get the host address fails do we set the host address to all zeroes to signify "failure" or "unkown" or do we do nothing and introduce a separate property for the request status?

nefarius, 2024-04-30:
Honestly I am displeased with the amount of workarounds currently required to send commands to the driver, maybe I should just add a vendor defined collection to each mode so we can send custom feature requests, like pairing.

PR #360 (2024-05-17): Rework pairing and host requesting
- Pairing now has 3 main modes:
    - Auto: to active host
    - Custom: to user defined Mac address
    - Disabled: never occurs
- Pairing can be triggered manually by enabling the new PairOnHotReload and hot-reloading configurations
    - Pairing is set to NOT occur if the desired new host is equal to the current set one, but I've included this option to prevent an edge case in which dshidmini could spam the controller on every hot reload with pairing requests if the driver failed to get the current controller's host
    - Resolves #315 
- DsHidMini now onl [... cut]
closed 2024-05-17 by nefarius